1a2m3/SPD-Reader-Writer #17: tested on H61 and H81 SMB "nothing found"

ankarle, 2022-03-21:
Hi, tested on H61 and H81 SMB I get on both "nothing found", is there anything I have to do or the chipsets aren't still supported?

1a2m3, 2022-03-21:
Are you running the program as admin?

ankarle, 2022-03-22:
yes I did run as admin, moreover I tried in the meantime on another two pcs with LGA1150, same result. 
All the pcs have windows 10 x64

1a2m3, 2022-03-22:
When you run the program, does event log show any driver info? 

[image]

Please also post a CPU-Z report by running report.cmd from the archive attached and paste the txt contents here (or on pastebin and post a link).

[CPU-Z.zip](https://github.com/1a2m3/SPD-Reader-Writer/files/8321873/CPU-Z.zip)

ankarle, 2022-03-22:
no, which driver should I install?

1a2m3, 2022-03-22:
The WinRing0 driver is loaded automatically when program is started as admin, no manual installation is needed. Make sure all winring0 dll and sys files are present in the same dir where the program is located. The program has to be run from a directory located on a local drive. Also some anti-cheat software might be blocking the driver from being loaded, so try disabling/uninstalling if you have any of those running.

ankarle, 2022-03-22:
I did again but same result, it doesn't load the driver

1a2m3, 2022-03-22:
Ok, that explains. Either your account type isn't administrator, or the program isn't detecting it properly. 
Try this and show me the log output.

[20220322-test2.zip](https://github.com/1a2m3/SPD-Reader-Writer/files/8323618/20220322-test2.zip)

ankarle, 2022-03-22:
on H61 doesn't start, I just get a black window

1a2m3, 2022-03-22:
Something is preventing from acquiring full administrative rights. The test2 build uses two different methods to checks for administrative privileges before loading the driver.

Make sure your account has full administrative rights. Open command prompt as administrator, enter command "**whoami /priv**" without quotes, and post the output.

ankarle, 2022-03-23:
Permission name Description Status
========================================= 
SeIncreaseQuotaPrivilege Adjust memory quotas for a process Disabled
SeSecurityPrivilege Manage monitoring and security protocols Disabled
SeTakeOwnershipPrivilege Take ownership of files and objects Disabled
SeLoadDriverPrivilege Load and remove device drivers Disabled
SeSystemProfilePrivilege Creating a profile of system performance Disabled
SeSystemtimePrivilege Changing the system time Disabled
SeProfileSingleProcessPrivilege Create a profile for a single process Disabled
SeIncreaseBasePriorityPrivilege Raise the scheduling priority Disabled
SeCreatePagefilePrivilege Create a page file Disabled
SeBackupPrivilege Back up files and directories Disabled
SeRestorePrivilege Restore files and directories Disabled
SeShutdownPrivilege Shut down the system Disabled
SeDebugPrivilege Debugging of programmes Enabled
SeSystemEnvironmentPrivilege Changing firmware environment variables Disabled
SeChangeNotifyPrivilege Omit scanning Enabled
SeRemoteShutdownPrivilege Force shutdown from a remote system Disabled
SeUndockPrivilege Removing the computer from the docking station Disabled
SeManageVolumePrivilege Performing volume maintenance tasks Disabled
SeImpersonatePrivilege Accept client identity after authentication Enabled
SeCreateGlobalPrivilege Create global objects Enabled
SeIncreaseWorkingSetPrivilege Increase working set of a process Disabled
SeTimeZonePrivilege Change time zone Disabled
SeCreateSymbolicLinkPrivilege Create symbolic links Disabled
SeDelegateSessionUserImpersonatePrivilege Retrieve identity token for another user in the same session Disabled

1a2m3, 2022-03-23:
Ok, your privileges are good.

Try this build and post results. It skips test for administrative rights and goes straight to SMBus initialization. 

(You can right click on any event log line and select "copy all lines..." to paste here).

[20220323-test3.zip](https://github.com/1a2m3/SPD-Reader-Writer/files/8330561/20220323-test3.zip)

ankarle, 2022-03-23:
unfortunately same as before, just a black window

1a2m3, 2022-03-23:
Try this, I've removed embedded manifest, run manually as admin.

If main window doesn't appear, post console window contents, otherwise post both event log and console contents.

[20220323-test4.zip](https://github.com/1a2m3/SPD-Reader-Writer/files/8330863/20220323-test4.zip)

ankarle, 2022-03-23:
just console windows showing 4 messages, the last one is Initializing SMBbus

1a2m3, 2022-03-23:
Added extra debugging info, try again.

[20220323-test5.zip](https://github.com/1a2m3/SPD-Reader-Writer/files/8331059/20220323-test5.zip)

1a2m3, 2022-03-23:
Post the full output, I need to see what happens before and after that line appears.

ankarle, 2022-03-23:
[image]
first run

ankarle, 2022-03-23:
[image]
second run, it doesn't do the same anymore

1a2m3, 2022-03-23:
Try this and post the results. It's part of original original WinRing0 source code bundle, it's modified to load the driver and display the results only. Run the exe as admin.

[WinRing0Test.zip](https://github.com/1a2m3/SPD-Reader-Writer/files/8331454/WinRing0Test.zip)

ankarle, 2022-03-23:
no results, no any windows gets open, running on the powershell also no output

1a2m3, 2022-03-24:
Issue identified, stay tuned for fixed test build.

1a2m3, 2022-03-24:
Try this. Should be working this time.

Restart your system before running this, in case previous builds caused SMBus to lock and you are seeing console window only or "nothing found" in the connect window.

Previous versions didn't work on non-English Windows. This build was tested on English and non-English versions of Windows 10 x86 and x64 with HM55 and NM10 based systems.

[20220323-test6.zip](https://github.com/1a2m3/SPD-Reader-Writer/files/8338020/20220323-test6.zip)

ankarle, 2022-03-24:
Hey, seems to work now, congratulation! :) I tried the reading function, I currently cannot test the writing function

[image]

1a2m3, 2022-03-24:
Great! I'll be closing this issue and will release an updated version shortly along with other fixes. If you discover another bug, open a new issue as this one is resolved.

closed 2022-03-24 by 1a2m3