1a2m3/SPD-Reader-Writer #29: SPD-RW GUI crash when dialog Select Device openned

Sergey1Fox, 2022-10-24:
Hardware: Dell XPS 9560, I7700HQ, chipset Kaby lake, HM175, DDR4

[image]

1a2m3, 2022-10-24:
1. Do you have any serial port devices present?

2. Run CLI version as admin with the following parameters and show me the output:

- spdrwcli.exe /find
- spdrwcli.exe /find arduino
- spdrwcli.exe /find smbus

Sergey1Fox, 2022-10-26:
1. Yes, two serial bluetooth ports (WIFI card Killer Wireless 1535)
2. First and second command were interrupted by CTRL+C after 1 minutes of waiting
 
[image]

1a2m3, 2022-10-26:
Thanks. Issue identified and replicated with Bluetooth COM port present.

Replace the DLL file with this one and try both GUI and CLI versions again and report the results.

[20221026HF.zip](https://github.com/1a2m3/SPD-Reader-Writer/files/9874369/20221026HF.zip)

Sergey1Fox, 2022-10-27:
Thank you. Now it works very well.

1a2m3, 2022-10-27:
Great. I'll post the fix as pre-release then until the full update is ready.

closed 2022-10-27 by 1a2m3