1a2m3/SPD-Reader-Writer #37: Driver use authorization.

CPUID2021, 2023-10-06:
Who authorized you to use the cpuz driver ?

1a2m3, 2023-10-06:
https://www.cpuid.com/terms-of-service.html

closed 2023-10-06 by 1a2m3

CPUID2021, 2023-10-07:
Precisely, our TOS does not allow unauthorized use as mentionned here :

 

« … provided that you do not (and do not allow any third party to) copy, modify, create a derivative work of, reverse engineer, reverse assemble or otherwise attempt to discover any source code, sell, assign, sublicense, grant a security interest in or otherwise transfer any right in the CPUID Services or remove or obscure the copyright notice or other notices displayed on the content. »

 

 

Why don’t you code your own driver ??

 

https://www.cpuid.com/terms-of-service.html

1a2m3, 2023-10-07:
How can I contact you to confirm you are from CPUID? 

I don't think the part you quoted applies to me. I didn't distribute any sources or modification of your work.

The first part of the sentence you quoted, starts with "You are hereby granted a personal, nonexclusive, nontransferable, revocable, limited license **to view, reproduce**, print, cache, **store and distribute content retrieved from our Site** via a generally available consumer web browser".

CPU-Z, which you do allow to download, use and distribute, comes with the driver, which automatically gets installed whenever CPU-Z is started. Anyone can download CPU-Z and use your driver when CPU-Z installs it in %temp% directory.

You also sell [CPUID SDK](https://www.cpuid-pro.com/products-system-information-kit.php), which also uses cpuz driver, but if you aren't happy with a private individual using your SDK in my free project, I can remove all references to cpuz, deal?

Why are you witch-hunting me? There are hundreds of other projects that use CPUID  SDK and CPUZ drivers, yet you only used your account to shame me, even though you've been here for 2 years. 😔

1a2m3, 2023-10-07:
Or would you rather allow me include a reference to CPUID SDK in my project?

I'm a hobbyist programmer, not a lawyer, so if you are from CPUID, tell me how I can make things right.

CPUID2021, 2023-10-07:
Hello Artem,

 

My name is Franck Delattre, I’m well from CPUID, you can contact me at ***@***.*** ***@***.***>  or ***@***.*** ***@***.***> .

 

There is a major difference between distribute cpuz (and its associated driver files) and publically publish the IO control codes. I guess you did not find them in the nature, but by reverse. Then the TOS applies.

 

But more important : by publishing the IO control codes as you did in CpuZ.cs, you allow anyone to use the cpuz driver in his own project. That is an open door to any malwares / cheat programs / whatever requires kernel fonctions.

The consequence is very predictible : MS will simply block the driver. They already did in the past, FYI. 

Why do you think WinRing0 can’t be used anymore ?

 

For you information, the SDK does not provide the customers with a direct access to the driver functions. Instead, they use high level fonctions in a DLL.

 

Considering your work, you’re 200% skilled to produce your own kernel mode driver. I know the signature is a major problem, but MS drastic requirements are not my fault or responsability.

By exposing my driver interface as you did, you provide MS with a good reason to block my driver.

 

I don’t see any other alternative than removing CpuZ.cs.

 

Thanks,

Franck

1a2m3, 2023-10-07:
Hi, your email address got filtered by github, so I can't see it.

If you want to email me, you can do so at:
artem
mikoulyak
com

You know where to put **@** and **.**

(I don't want to word argue, but there are ways to find IOCTL codes, other than disassembly. Anyone can hook DeviceIoControl API to see what parameters are being passed to the driver handle.)

I am aware of cpuz driver being blacklisted in the past, but that was because it allowed arbitrary memory writing, and it didn't check access rights, and it allowed multiples handles to be opened. Well, if you coded cpuz driver, then you should know what was wrong with cpuz138, cpuz143 and earlier versions, and how it was fixed in future versions. 

I understand your concern, so I removed CpuZ.cs class from all current branches, because I believe it is within everyone's interests and benefit, when software is not used with malicious intent.

Can we consider this matter resolved?