1a2m3/SPD-Reader-Writer #39: question about the eda picture

right-0903, 2023-11-02:
I should state this at first, my knownledge about circuit is very limited. But I want to remove rswp in my so-dimm, I think I just need to know how to get the wires connected correctly, so I look into these pins. I have a glance at `first-prototype.jpg` in the `images` folder and the eda in the `schematic` folder. The questions are

1. Sould I really need a `C1` container? I don't notice that the container is connected to the circuit.
2. Which resistors are unnecessary? Resistor R1 R2 and so on, they are not connected to the circuit too.
3. If SA1_CTL is necessary for ddr4? you don't talk about SA1_CTL in evga thread.
> SA1_CTL is used to switch pin SA1 on EEPROM between VCC and ground. This is needed because RSWP-supported EEPROMs used on DDR3 and DDR2 require different SA1 configurations to enable or disable RSWP. It is also used to prevent addressing conflicts when DDR5 is used.

1a2m3, 2023-11-02:
Hi,

The first prototype didn't have any write protection capabilities and is no longer supported by firmware and software. (It wasn't built according to the current schematic, it was just a piece of PCB cut out of an old motherboard.) Refer to the latest [schematic](https://github.com/1a2m3/SPD-Reader-Writer/blob/master/schematic/SpdReaderWriterSchematic.pdf). 

I'll answer your questions in regards to the latest schematic.

1. The C1 capacitor isn't necessary, it is there for power stability. You will most likely be fine without it.
2. Actually R1 and R2 pull-ups can be omitted, but you might run into issues when running I2C in fast mode. All other resistors in their modules are mandatory as they serve current limiting and voltage divider purposes.
3. Thanks for noting, the guide has been updated. The missing pin was pin 78 on DDR4 UDIMM.
4. If you only need DDR4 support, then you can get away without SA1_CTL module. SA1_CTL serves two purposes: provides RSWP support for DDR3 and older RAM, and prevents addressing conflicts when DDR5 is used, as DDR5 is permanently on address 80, and SA1_CTL keeps the other RAM at address 81.

right-0903, 2023-11-03:
@1a2m3 OK, thanks. another question, you soldering pin 223 VDD, but you don't mention this, you show us `gnd=2*`, shoud pin 233 be connected to GND?

1a2m3, 2023-11-03:
No, pins 223 and 233 are not used. Those are VDD pins, required for RAM operation, but not for the programmer. The only power pin you need to use is VDDSPD. For ground you can use any of the VSS pins.

right-0903, 2023-11-03:
@1a2m3 Sorry 23 is a typo, I mean 223. I got this from micron 
> pin 4 Vssspd Supply Power
> pin 8 VDDSPD Supply Ground

it says vddspd is gnd, btw, can you show me all necessary soldering pins? I am asking my friend to solder. thanks.

1a2m3, 2023-11-03:
All necessary pins you need are mentioned in the guide with photos on the [forum here](https://forums.evga.com/m3053544.aspx).

1a2m3, 2023-11-03:
> pin 4 Vssspd Supply Power
> pin 8 VDDSPD Supply Ground

Omg, They mixed those up, VSS is ground, VDD is power. 
Pin 8 on EEPROM should be VDD, ground is pin 4.

[This document](https://www.onsemi.com/pdf/datasheet/n34c04-d.pdf) shows correct EEPROM pin descriptions.

right-0903, 2023-11-03:
> [This document](https://www.onsemi.com/pdf/datasheet/n34c04-d.pdf) shows correct EEPROM pin descriptions.

I see. The materials are shipping, I will try it later. Thanks again!

right-0903, 2023-11-06:
@1a2m3 I forget about 3.3k resistor, now I have only 1k, 510, 5.6k, can I use 3k,4k,or 5.6k to replace it?

1a2m3, 2023-11-06:
The 3.3K resistor (R5) is part of the voltage divider loop for a VHV feedback circuit. It's value is tied to the R4 resistor, so you can't just replace it alone. It has to be ~60% of R4's value. The values I chose (5.6K and 3.3K) will result in 3.3V at the feedback output, which works for either 3.3V or 5V arduinos. You are free to adjust R4 and R5, just make sure the feedback voltage is within your arduino's safe input voltages (that can be 5V or 3.3V, depending on model), when HV_EN signal is on.

1a2m3, 2023-11-06:
1K won't work.
3K might be a little too low for 5V models and is fine for 3.3V models. 
4K is fine for 5V models, but might be unsafe for 3.3V models.
5.6K is fine for 5V models, but is not safe for 3.3V models.

right-0903, 2023-11-06:
> The 3.3K resistor (R5) is part of the voltage divider loop for a VHV feedback circuit. It's value is tied to the R4 resistor, so you can't just replace it alone. It has to be ~60% of R4's value. The [...]

inspiring answer, I use serial 6.6k and serial 4k to achieve it.

right-0903, 2023-11-06:
nice work!

closed 2023-11-06 by right-0903