1a2m3/SPD-Reader-Writer #43: Cant disable RSWP on AM4

WewerRaver, 2024-01-21:
Hi 1A!
Great thanks for your tool!!!
I find Your thread on EVGA forum ("My Arduino-based RAM SPD reader/writer") but site looks like for americans, so i don't register there
I'm from Ukraine, so, maybe my english not very well, sorry, Will believe in Google-translate) and hope readers can understand message.
I find B250 Biostar mobo some days ago and after short bios-mod it become SPD-via-SMBus rewritable)
But, 
Problem #1: RSWP, all 3 SA-lines shorted to GND
Problem #2: I soldered a RAM-half slot extension(like in programmer), which include all 8 pins for SPD chip in parallel to DIMM2 slot
The first time the wires were about 20 cm long, but the program did not see any in-slot inserted sticks.
However, if I hot-plug sticks into the board itself, they are  detected correctly.
At now wires shorter than 10cm, max close to board but situation the same, maybe board want some signal to activate slot, like termination or something else?

This is the background..)
Today I found a new board MSI B450-A PRO MAX, good am4 mobo with 4 dimm slot
Board is very interesting for this task
In slot A1 and B1, SA0 lines are shorted to ground like B250, but A2 and B2 are connected to 3.3V via 1kΩ separately from each other.
This is a triumph, I thought!
I quickly soldered the 3.3/9v SA0 jumper, the 9V battery has been there since the B250)
Starting Windows, launching SPDRWgui,,, 

Problem #3: Long initialization time, more than a minute, then connecting - the same thing 1+min
Perhaps this is a feature of the ZEN board or CPU 200GE..

Problem #4: Press F6 - yes, 3 of 4 block a RSWP Protected(Micron)
Change My jumper to 9v(tester show 9.78v), click on Clear RSWP icon and after few seconds - Cleared Ok!
And this is happiness!!! )))
Press F6 for checking - 3 of 4 block a RSWP Protected!?!?!? ((( maybe reboot needed!?!  I understand that no but
Rebooting system even not changing jumper position, Windows, SPDRWgui, F6,, same,, 3 of 4 protected
Trying with different sticks(Samsung, Hynix) - any changes. 

So, not briefly, but I tried to describe my actions in detail
At a minimum, the B450-A PRO MAX can make a good home system and a good programmer for 2 sticks at once)

Regards

1a2m3, 2024-01-21:
Ок, здесь можно писать на русском или украинском.

RSWP через SMBus можно поставить только на платах с DDR5. Снять защиту RSWP без модификаций можно на некоторых платах Asus серии Maximus, у которых есть разъем [MR_TEST2](https://forums.evga.com/FindPost/3616968). В остальных случаях статус RSWP можно только прочитать через SMBus. 

Для полноценной поддержки RSWP лучше собрать программатор на Arduino.

Когда подавал 9V на SA0, сколько модулей памяти было на материке в момент попытки снятия защиты?

WewerRaver, 2024-01-21:
Привет Артем!
Я - Дима
Та я плясал с єтим бубном по разному 1,2,3 )

WewerRaver, 2024-01-21:
Странно конечно, а какая этой смбусе разница?
Есть параметры микрули, она же их поддерживает
В250ю, что я разлочил, плашки, где нет защиты - она пишет норм, АМ4 не
проверял но думаю также а значит параметрически есть полное соответствие
Вопрос снять РСВП для 99% плат - подача 9в на ногу, понятно, в слоте так не
получиться, да и в принципе нужно либо каскад замутить с 12в к 9в, как на
упомянутой Вами маме скорее всего сделано, с этим джампером специфическим
ибо  шибко не стандартная напруга для компа, либо как я, устраивать
аттракцион "Очумелые ручки"
И вот, железячно условия соблюдены, адресное пространство - раз
пишет-читает в чем тут может быть проблема?
Ну и конечно странно, что "внешний порт" не читает

WewerRaver, 2024-01-21:
[image]

1a2m3, 2024-01-22:
Я не проверял работу RSWP на DDR4 через SMBus, так что подтвердить работоспособность не могу, но команда снятия защиты одинаковая на SMBus и Arduino. 

Условия для работы RSWP на DDR5 отличаются от RSWP на любой другой памяти, (для DDR5 надо SPDhub в режим offline перевести, замкнув вывод HSA на землю, что, скорее всего, и делает перемычка MR_TEST2).

Попробуй так - установи только один модуль, предварительно заизолировав контакт SA0 на планке или слоте, подай 9V на ногу SA0 у EEPROM и попробуй выполнить очистку RSWP. Если не будет загружаться комп, то подай 9V после того, как POST будет пройден.

По поводу долгой инициализации - для начала покажи текстовый отчет CPU-Z. Запусти последнюю версию CPU-Z с дефолтными параметрами и нажми Tools > Save report as .TXT. Прикрепи файлом к ответу.

И попробуй консольную утилиту и скажи как быстро выводится результат. Выполни команду из-под админа:
`spdrwcli.exe /find smbus`

WewerRaver, 2024-01-22:
Именно поэтому я хвалил мсю, если в биостаре жестко все на земле, от
контакта SA никаких дорог, звониться на 0 в короткую и там в принципе
наверное не вариант как-то разлочить, разве контакт из слота выдрать,
потому я и сделал выносной кусок слота, как на фото
То вот у мси от ноги SA0 дорога через резюк 1кОм на 3,3в сидит, и там уже
сделан джампер)  3,3в<SA0>9v (крона, "-" соединен с землей мамы, все
по-феншую)
И такой потанцевал для 2х слотов А2 и В2!!!
Сначала я стартанул систему обычно(с одной плашкой), просто проверить,
получил такую же фигу как и на штеуде, прям ругнулася прога, мол фу-фу-фу,
дуля тобі і жовте віконечко!. Что интересно, Н170, В250 - там вся логика
SA0-2 на земле, завтра еще высус prime Z270-A гляну, хотя , думаю, будут те
же яйца, токо в прохвиль, а вот у этой мси - наоборот - SA0,1 на +3,3v
заведены SA2 не звонил но думаю та же песня. Я даташиты на несколько типов
этих микрух изучил, понятно, они аналогичны, там триггер для снятия это Hv
on SA0 ну и  команда с программатора. В принципе, любая мать могла б
писать-читать, но "материнщики" договорились с "памятниками" и в итоге 1й
блок на запись на уровне биоса , 2й - вот этот Hv on SA0. Сейчас этой инфы,
если толково - не очень, а лет 5-6 тому, когда ее актуальность была более
чем.. разве за большие деньги и то только китайцам)

Следующий этап - переключил SA0 на 9в, результат как написал: прога все
сделала написала мне ок но перепроверка показала те же 3 блока из 4х в
защите
Следующее - перегрузка при 9в ну и новая попытка, далее - поскольку
подключение реально долгое, в отличие от того же интела, где пару сек,
сделал clear раз 5-6
Дальше - поставил 2-ю планку, тут вообще чудеса, "робочая" стояла как пишет
мать, в В2, доставил в А2, из нее читает какую-то дичь, я потом отдельно
даже с нее стартовал и тоже самое, кину скрин завтра, вместе с отчетом
мне вот интересно, почему в250я не видит внешнюю плашку..
попробую завтра взять какую-то дохлую плашку и с нее сдую SPD, втыкну в
слот мамки а вторую поставлю на удлинитель, может так увидит

Благодарю за ответ
на/до связи

WewerRaver, 2024-01-22:
Приветствую

CPU-Z TXT Report
-------------------------------------------------------------------------

Binaries
-------------------------------------------------------------------------

CPU-Z version			2.08.0.x64

Processors
-------------------------------------------------------------------------

CPU Groups			1
CPU Group 0			4 threads, mask=0xF

Number of sockets		1
Number of threads		4

APICs
-------------------------------------------------------------------------

Socket 0	
	-- Node 0	
		-- CCX 0		
			-- Core 0 (ID 0)	
				-- Thread 0	0
				-- Thread 1	1
			-- Core 1 (ID 1)	
				-- Thread 2	2
				-- Thread 3	3

Timers
-------------------------------------------------------------------------

	ACPI timer		3.580 MHz
	Perf timer		10.000 MHz
	Sys timer		1.000 KHz

Processors Information
-------------------------------------------------------------------------

Socket 1			ID = 0
	Number of cores		2 (max 2)
	Number of threads	4 (max 4)
	Number of CCDs		0
	Manufacturer		AuthenticAMD
	Name			AMD Athlon 200GE
	Codename		Raven Ridge
	Specification		AMD Athlon 200GE with Radeon Vega Graphics     
	Package 		Socket AM4 (1331)
	CPUID			F.1.0
	Extended CPUID		17.11
	Core Stepping		RV-B0
	Technology		14 nm
	Core Speed		1595.9 MHz
	Multiplier x Bus Speed	16.0 x 99.7 MHz
	Base frequency (cores)	99.7 MHz
	Base frequency (mem.)	99.7 MHz
	Instructions sets	MMX (+), SSE, SSE2, SSE3, SSSE3, SSE4.1, SSE4.2, SSE4A, x86-64, AMD-V, AES, AVX, AVX2, FMA3, SHA
	Microcode Revision	0x8101016
	L1 Data cache		2 x 32 KB (8-way, 64-byte line)
	L1 Instruction cache	2 x 64 KB (4-way, 64-byte line)
	L2 cache		2 x 512 KB (8-way, 64-byte line)
	L3 cache		4 MB (16-way, 64-byte line)
	Max CPUID level		0000000Dh
	Max CPUID ext. level	8000001Fh
	FID/VID Control		yes
	# of P-States		3
	P-State			FID 0x880 - VID 0x28 (32.00x - 1.300 V)
[... 1535 lines cut ...]
	Caps class		Power Management
	Caps offset		0x78
	Caps version		1.2
PCI capability
	Caps class		PCI Express
	Caps offset		0x80
	Device type		Upstream Port of PCI-E Switch
	Port			0
	Caps version		2.0
	Link width		2x (max 4x)
	Link speed		8.0 GT/s (max 8.0 GT/s)
PCI capability
	Caps class		Subsystem Vendor
	Caps offset		0xC0
	SubVendor ID		0x1B21 (ASMedia)
	SubSystem ID		0x0201
PCI extended capability
	Caps class		Advanced Error Reporting
	Caps offset		0x100
PCI registers	
		00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 
	00	22 10 C6 43 07 04 10 00 01 00 04 06 10 00 81 00 
	10	00 00 00 00 00 00 00 00 12 20 25 00 F1 F1 00 00 
	20	D0 FC D0 FC F1 FF 01 00 00 00 00 00 00 00 00 00 
	30	00 00 00 00 50 00 00 00 00 00 00 00

1a2m3, 2024-01-22:
Прикрепи файлом, отчет обрезан. Не отвечай через email, зайди на github и ответь через форму в теме, нажав Paste, drop, or click to add files.

WewerRaver, 2024-01-22:
[WIN-TOBHTGBS2KA.txt](https://github.com/1a2m3/SPD-Reader-Writer/files/14009209/WIN-TOBHTGBS2KA.txt)

WewerRaver, 2024-01-22:
Отправил

1a2m3, 2024-01-22:
Ок, в настройках попробуй поставить Maximum PCI bus = 1 (Settings > Hardware > Smbus). И перезапусти SPD-RW.

А также запусти  `spdrwcli.exe /find smbus` из под админа, как быстро выводится результат?

WewerRaver, 2024-01-22:
И попробуй консольную утилиту и скажи как быстро выводится результат.
Выполни команду из-под админа:
spdrwcli.exe /find smbus - вообще нет
cpuz158 initialization failure

1a2m3, 2024-01-23:
Что-то не дает запуститься драйверу. Хотя раз CPU-Z сработал, то сам драйвер работает нормально.

Запусти этот тест от админа и покажи результат. Это консольная программа, так что лучше запусти сначала cmd.exe от админа и из нее уже запусти SpdRwTestApp.exe

[20240122-test.zip](https://github.com/1a2m3/SPD-Reader-Writer/files/14017601/20240122-test.zip)

WewerRaver, 2024-01-23:
Кстати, версия на "ринге" 20230205 норм запускается но там нет возможности снятия RSWP для SMBus и читает она инверсно, из 4 блоков 1 закрыт а 3 доступно)

WewerRaver, 2024-01-23:
нашел причину!
я для удобства ехе-шник переименовал, вернул название и все заработало)
определяется +/- также 45-50сек, шить не дает, говорит что не сом-порт

1a2m3, 2024-01-24:
Покажи лог и скриншот ошибки.

WewerRaver, 2024-01-24:
[image]

конечно же, все на админ-правах

1a2m3, 2024-01-24:
У тебя неправильный синтаксис и команда `/disablewriteprotection` предназначена только для Arduino. 

Давай лог из тестовой проги и GUI.

1a2m3, 2024-01-25:
Лог в GUI отображается во вкладке "Event log", сохранить можно вручную, нажава "Save to file...".

Тестовая - без ключей.

WewerRaver, 2024-01-25:
мин 10

WewerRaver, 2024-01-25:
C:\Users\user\Desktop\spd\20240122-test>C:\Users\user\Desktop\spd\20240122-test\SpdRwTestApp.exe
Driver.Start
ServiceName: cpuz158
DeviceName: \\.\cpuz158
FileName: C:\Users\user\AppData\Local\Temp\cpuz158
FilePath:
Driver.Stop

C:\Users\user\Desktop\spd\20240122-test>

WewerRaver, 2024-01-25:
[Info] 25.01.2024 0:24:21: Program started (v. 2.23.12.10, Core: 2.23.12.10 ) (Included FW: 20231207)
[Info] 25.01.2024 0:24:21: Driver installed: cpuz158 (v.1.0.5.8)
[Info] 25.01.2024 0:26:15: AMD ZEN (0): Connecting to SMBus
[Info] 25.01.2024 0:26:15: AMD ZEN (0): Ready!
[Info] 25.01.2024 0:26:23: 1022:790B: Reading from 1022:790B:83
[Error] 25.01.2024 0:26:23: Read error: AMD ZEN:83:353
[Error] 25.01.2024 0:26:27: 1022:790B: Unable to read data from 1022:790B
[Info] 25.01.2024 0:26:31: 1022:790B: Reading from 1022:790B:83
[Info] 25.01.2024 0:26:32: 1022:790B: Read 512 bytes from 1022:790B:83 in 0,34 sec.

WewerRaver, 2024-01-25:
"У тебя неправильный синтаксис и команда /disablewriteprotection предназначена только для Arduino"
- использовал что есть, других в принципе нет), потому и говорю для смбуса не работает cli

1a2m3, 2024-01-25:
Скорее в фоне работает софт, который обращается к SMBus, из-за чего происходят ошибки чтения. И отсутствие пути после FilePath тестовой проги говорит о том, что драйвер cpuz158 уже присутствует. 

После закрытия SPD-RW, запусти из-под админа remove_driver.cmd из папки tools. Если будет сообщение, что драйвер удален, то значит в фоне присутствует приложение, использующее тот же драйвер, которое переустанавливает его.

Если драйвер продолжает работать, то закрой софт мониторинга, управления подсветки итд, до тех пор, пока remove_driver.cmd не сообщит, что драйвера нет. Потом попробуй чтение еще несколько раз.

А долгая загрузка на каком этапе? Между запуском и появлением окна программы? По логу критических задержек нет.

WewerRaver, 2024-01-26:
oke, наверное вечером завтра продолжу

WewerRaver, 2024-01-29:
Здравствуй Артем, вот, только появилась минутка
моя "песочница" не подключена к нету, возможно потому последний этап не прошел успешно и выдало:
-- Message #0
-----------------------------------------------------------------

Instrumentation was requested, but wasn't made

-----------------------------------------------------------------
однако оно сейвится в в локалах 
папка сессии оттуда

WewerRaver, 2024-01-29:
[spdrwcli.exe 240129_125130870.zip](https://github.com/1a2m3/SPD-Reader-Writer/files/14078587/spdrwcli.exe.240129_125130870.zip)

1a2m3, 2024-01-29:
Спасибо за отчет, открыть удалось, буду изучать.

1a2m3, 2024-01-29:
Попробуй `/find smbus` с этой версией. Если будет опять долго, то сделай отчет в dotTrace, как и ранее.

[20240128-2.zip](https://github.com/1a2m3/SPD-Reader-Writer/files/14079037/20240128-2.zip)

WewerRaver, 2024-01-29:
Hello!)
A можно добавить пару строк и команду для переадресации в CLI с СОМ to SMBus?
Чтоб уже All inclusive был)

1a2m3, 2024-01-30:
Давай завершим разбирательство с текущей проблемой, а для пожеланий создай новый запрос.

WewerRaver, 2024-01-30:
Привет, текущая проблема - это долгий детект на АМ4, верно?
Я сегодня сделаю в 240129

WewerRaver, 2024-01-30:
"Ссылка на обьект не указывает на екземплар обьекта"
выдает так, причем даже когда ищу типа ардуину, просто /find
где-то указатель видимо или не тот или не туда)

1a2m3, 2024-01-31:
Это с какой командой? Покажи ошибку полностью и давай отчет.

WewerRaver, 2024-01-31:
spdrwcli /find
spdrwcli /find smbus

без разницы
и красным 
"Ссылка на обьект не указывает на екземплар обьекта"

WewerRaver, 2024-01-31:
нечего показывать)

1a2m3, 2024-01-31:
Попробуй с этим билдом. Ничего не менял по сравнению с предыдущим, просто это debug билд и ошибки будут подробнее описаны, если таковые будут. Если ошибок не будет, то сразу сделай отчет.

[20240128-d.zip](https://github.com/1a2m3/SPD-Reader-Writer/files/14107259/20240128-d.zip)

WewerRaver, 2024-02-01:
[image]

WewerRaver, 2024-02-01:
думаю, понял причину, почему не снимало
Исходя из таблицы кроме , 9в на SA0-м пине должны также быть сконфигурированы SA1, SA2
SA1 еще нужно на 3.3в переключить, и, возможно, будет мне счастье!)

1a2m3, 2024-02-01:
Пробуй эту версию.

[20240131-d.zip](https://github.com/1a2m3/SPD-Reader-Writer/files/14120254/20240131-d.zip)

1a2m3, 2024-02-01:
Конфигурация SA1 требовалась для работы RSWP на DDR3-2, а SA2 вообще не играет роли, тем более, что не на всех модулях он выведен. Официально, в соответствии со стандартами Jedec, для RSWP на DDR4 требуется только 9V на SA0, но по сообщениям других пользователей, им встречались модули DDR4, на которых EEPROM частично требовали выполнения условий, необходимых для DDR3 работы с RSWP, касательно конфигурации SA1.

WewerRaver, 2024-02-01:
так отож, кто его, єто амудє знает, нужно попробовать

WewerRaver, 2024-02-01:
жду еще один твой "релиз", будем тестить далее)

1a2m3, 2024-02-01:
А чего последний тестируешь? Я исходя из результатов вношу изменения. Какие изменения ты ожидаешь, если я не знаю, есть ли результат от предыдущих?

WewerRaver, 2024-02-02:
по версии d я бросил скрин, или запускать в доттрейсе не смотря на сообщения об ошибках?
прога не выдала результат и я подумал что она сбоит

1a2m3, 2024-02-02:
Никаких скринов и сообщений по версии [20240131-d.zip](https://github.com/1a2m3/SPD-Reader-Writer/files/14120254/20240131-d.zip) нету.

WewerRaver, 2024-02-02:
[image]

WewerRaver, 2024-02-02:
вот, повторно

1a2m3, 2024-02-02:
[20240202-d.zip](https://github.com/1a2m3/SPD-Reader-Writer/files/14146221/20240202-d.zip)

Давай с этой пробуй.

Добавил PDB для более детального отображения ошибки, не могу понять, что там в Initialize() может быть Null'ем.

WewerRaver, 2024-02-04:
[image]
привет, в доттрейсе запустить?

1a2m3, 2024-02-04:
Ок, понял в чем дело, исправил NullReferenceException.

Пробуй: [20240204-d.zip](https://github.com/1a2m3/SPD-Reader-Writer/files/14158676/20240204-d.zip)

Если `/find smbus` выдаст ошибку, то давай ее сюда. 
Если выдаст результат без ошибки, но долго будет искать, то давай отчет в dotTrace.

WewerRaver, 2024-02-05:
как-то так и видел, потому и переспросил нужно ли с еророй ее дотить&трейсить)

WewerRaver, 2024-02-10:
Здравствуйте уважаемый!
только вот руки с мозгами дошли до проверки
1. Работает
2. Значительно быстрее секунд 15, что в сравнении с предыдущим, конечно, очень ощутимо

WewerRaver, 2024-02-10:
счас в "песочницу" закину и вышлю результат

WewerRaver, 2024-02-10:
[spdrwcli.exe 240210_041545383.zip](https://github.com/1a2m3/SPD-Reader-Writer/files/14229846/spdrwcli.exe.240210_041545383.zip)

1a2m3, 2024-02-11:
Ок, проверяй этот билд. Сделать надо то же, что и в прошлый раз.

[20240210-d.zip](https://github.com/1a2m3/SPD-Reader-Writer/files/14231187/20240210-d.zip)

Уменьшил количество вызовов GetBusStatus в обеих WaitForStatus, что существенно должно уменьшит вызовы ReadPortEx и убрал ненужные проверки каждый раз при вызове ApiQuery.

WewerRaver, 2024-02-12:
[dotTraceSnapshots.zip](https://github.com/1a2m3/SPD-Reader-Writer/files/14252711/dotTraceSnapshots.zip)

Похоже не там смотришь, результат тот же 15-18сек

1a2m3, 2024-02-13:
По прежнему больше всего времени занимает ожидание статуса (WaitForStatus). Количество вызовов меньше стало, но это за счет Sleep в цикле ожидания.

Закинь этот файлик в директорию с последней версией ([20240210-d.zip](https://github.com/1a2m3/SPD-Reader-Writer/files/14231187/20240210-d.zip)), и выполни его в cmd, запущенной от администратора. Когда увидишь "done!", полученный файл (smb_output.txt) приложи к ответу.

[smbusreadtest.zip](https://github.com/1a2m3/SPD-Reader-Writer/files/14255023/smbusreadtest.zip)

WewerRaver, 2024-02-13:
амд может сегодня или уже завтра сделаю
разобрался с интелом - плашки "ловит" на выносном сокете, том, что в самом начале фотка, но, увы, та же песня, все тот же биостар В250 тоже уже типа снимает защиту но не снимает(((

WewerRaver, 2024-02-13:
Пробовал hynix, micron, samsung, у них где 2 где 3 странички в блоке но УВЫ
Счас пока штеуд на стенде - попробую с SA1-2 поиграться а там как сил хватит, почти 5 утра уже)

[30 more replies]

closed 2024-06-23 by 1a2m3